pc2/sus-compiler #3: `if` and `when` for compiletime or runtime conditions

VonTum, 2024-08-12:
We should use an extra keyword `when` to distinguish runtime conditions from compiletime codegen conditions? 

There's a semantic difference between them. `if` prevents a section of hardware from being generated, while `when` just adds runtime conditions to it. 

The big benefits are clarity of intent on the programmer's part. Also the benefit is the compiler can warn the programmer when using a `when` with a compile time constant, or an error on `if` with a non-constant. 

The rest of the syntax can be entirely shared!

TomSkarabis, 2024-12-18:
First half is here: https://github.com/pc2/sus-compiler/commit/ceeccc696f6bf49ed32dd2f75448e90372a65fc9 , if you want to take a look at it @VonTum . The insane amount of line changes are because I had to regenerate the tree-sitter stuff. Is this normal or did I break something?

 I will start a merge request once I get the warnings to work.

VonTum, 2024-12-20:
I've created a draft pull request: https://github.com/pc2/sus-compiler/pull/43

I'll do my code review on there

closed 2025-01-30 by VonTum