pc2/sus-compiler #23: VHDL Backend

VonTum, 2024-11-09:
Allow outputting VHDL code too

IBims1NicerTobi, 2024-11-12:
I would like to do this

VonTum, 2024-12-03:
@IBims1NicerTobi I would like to implement #25 such that Tim & Fabian can implement #27 more comprehensively. This would change the Verilog backend, as well as influencing the VHDL backend afterwards. Is that alright with you?

IBims1NicerTobi, 2024-12-03:
Sure but should I then just wait until #25 is done so I don't have to implement this twice? There are other things I could work on in the mean time and I would prefer that to doing it twice.

VonTum, 2024-12-03:
I wouldn't wait to finish the VHDL backend, as it doesn't change much. It's just a special case for _if_ the submodule is an `extern module`

IBims1NicerTobi, 2024-12-03:
Sure

VonTum, 2024-12-03:
I estimate it'll take me a few days to get to it too, since I'm first setting up sus.rocks