pc2/sus-compiler #64: CSE when assigning non-gen value to (valid) array access of gen

pbeart, 2025-02-12:
E.g. this works:
```gen int[2] p
gen int q = p[0]```
so does this:
```int[2] p
int q = p[0]```

but this gives a CSE:

```gen int[2] p
int q = p[0]```

The error is:
```thread 'main' panicked at src\value.rs:55:44:
internal error: entered unreachable code
stack backtrace:
   0: std::panicking::begin_panic_handler
             at /rustc/e71f9a9a98b0faf423844bf0ba7438f29dc27d58\library/std\src\panicking.rs:665
   1: core::panicking::panic_fmt
             at /rustc/e71f9a9a98b0faf423844bf0ba7438f29dc27d58\library/core\src\panicking.rs:76
   2: core::panicking::panic
             at /rustc/e71f9a9a98b0faf423844bf0ba7438f29dc27d58\library/core\src\panicking.rs:148
   3: enum2$<sus_compiler::value::Value>::get_type_best_effort
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\value.rs:55
   4: sus_compiler::instantiation::InstantiationContext::alloc_wire_for_const
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\execute.rs:507
   5: sus_compiler::instantiation::InstantiationContext::get_wire_or_constant_as_wire
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\execute.rs:528
   6: sus_compiler::instantiation::InstantiationContext::instantiate_connection
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\execute.rs:379
   7: sus_compiler::instantiation::InstantiationContext::instantiate_code_block
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\execute.rs:766
   8: sus_compiler::instantiation::InstantiationContext::execute_module
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\execute.rs:908
   9: sus_compiler::instantiation::perform_instantiation
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\mod.rs:405
  10: sus_compiler::instantiation::InstantiationCache::instantiate
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\instantiation\mod.rs:240
  11: sus_compiler::linker::Linker::recompile_all
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\compiler_top.rs:239
  12: sus_compiler::linker::Linker::update_text
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\dev_aid\lsp\mod.rs:84
  13: sus_compiler::dev_aid::lsp::lsp_main
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\dev_aid\lsp\mod.rs:633
  14: sus_compiler::main
             at C:\Users\verya\Projects\sus_clean\sus-compiler\src\main.rs:49```

Also worth noting that if the index is out of bounds, no CSE occurs (e.g. the following code does *not* trigger a CSE despite following the same pattern of non-gen variable using array access of a gen variable)

```gen int[2] p
int q = p[3]```

pbeart, 2025-02-17:
More information: this also occurs when assigning an array to the value of a gen array, correctly sized or not, i.e. all of the following cause ICE:
```
gen int[3] a
gen int[3] b = a
```

```
gen int[3] a
gen int[2] b = a
```

```
gen int[3] a
int[2] b = a
```

VonTum, 2025-07-15:
Fixed in master