pc2/sus-compiler #88: SystemVerilog-style slices of arbitrary base

VonTum, 2025-07-19:


muinmist, 2025-08-02:
Hi. Is this issue kind of self-explanatory or will you post a description later? I am new to github contribution and this question could as well be a silly one. Thanks.

VonTum, 2025-08-02:
Hi @muinmist, I'll admit this issue is a bit low in detail. 

What it describes is extending the way SUS arrays work, such that one can pick an arbitrary range for the array indices - like in systemverilog. 

int[5] has the valid range of indices: 0-4 inclusive. 
But we want to extend the language with say int[3:8], to again have a 5-element array, but with indices 3-7 inclusive. 

The reason we want this is that we don't want to force an additional subtract unit if the user's design doesn't need it - a hardware multiplexer doesn't care if its using 00,01,10 as selector, or 01,10,11 for instance